dalathegreat/Nissan-LEAF-Battery-Upgrade #1: ZE0 - 80% timer does not work

dalathegreat, 2023-06-22:
The dashboard 80% timer does not work on the 2011-2012 ZE0. The newer battery packs don't understand the 80% longlife option at all. However, it would be possible to code this into the CAN-bridge. The 0x1F2 message sent by the VCM contains a parameter called "TargetChargeSOC". This bit has two values, 0=100% and 1=80%. In theory, we could quite simply check if this is active, and force a charge stop when 80% is reached. Some coding needed ofc :)

malyjaponec, 2024-03-05:
This might help: in the version of CAN-bridge (3 port) ti worked in my ZE0 in version v.2.54. It was not exactly 80% it usually stopped at 81% but never happened that it chargers over this level.

I am ready to test a special release I am charging car from 70 to 80% every day or I was before updating to latest version.

malyjaponec, 2024-03-15:
I do not know if it should have separate issue but the car after upgrade from v 2.54 to latest lost ability to predict charging time on 110V it if there is 2 hours on 220V it shows like 13+ hours on 110V that is completely nonsense. Also and what is more important it lost ability time correctly charging. Even saying 1:40 minutes to 80% or 2:40 to 100% it with timer set starts to charge immediately as plug is inserted in most of times. Time to time it delays the charging but I did not find any pattern. Looking on yesterday data it end night charging at 5:15 at 80 stopped by OVMS when it was scheduled to finish (100%) at 6.30. If it was not stopped by OVMS that limits SOC it would probably timed well, but now if I plug it it starts at the moment and basically I did nothing with car timer. 
There are some packet translations that were in 2.54 that are for ZE0 essential that are missing in latest software. That is where I would start to investigace first.

malyjaponec, 2024-04-03:
I went back to 2.54 recently and the expected times for charging 220 and also 110V seems OK again. It really looks that old software was able to compute time correctly and also charging timer worked. With latest can bridge sw the car starts to charge immediately even it needs 2-3 hours and end time is 15 hours ahead. Both (80% stop and timing of charge end) was already working in old version.

OkinawaJapan, 2024-04-03:
> I went back to 2.54 recently and the expected times for charging 220 and also 110V seems OK again. It really looks that old software was able to compute time correctly and also charging timer [...]

Very interesting. Where do I get the older 2.54 version? I don't see it listed anymore? And is that for the 2-port can-bridge?

malyjaponec, 2024-04-03:
I do not have 2.54 source. I have only binary from Dala. Also it was the sw I originally got in the can bridge. I am using 3 port hardware and car is ZE0 with 30kWh battery.
I am not sure if this old sw has public code. If so I think i would be relatively simple to find out how the packet were made in that ancient version and correct computing times end "80% end" in current code.

malyjaponec, 2024-04-04:
I can confirm that car charging timer works again after returning to firmware 2.54. It delays charging to time before requested "finish time". It did it little bit early. It is normal. Probably the estimations are more then little bit pessimistic. I will try to compare estimation on display and what is real time if i remember.

I consider that this issue is **not only about 80% stop now but also about charging timing** because if the time estimation is wrong also 100% charging timer does not do what is expected.

When I tested 4.20 the estimations was crazy especially for 110V. So it seems that the cause that car do not want to postpone charging is probably caused by incorrect time estimations also. It just my guess.

_If somebody wants to solve this problem and makes some beta version that used 2.54 way of computing time estimations and 80% estimation I am ready to test. I would like to try it but my knowledge of CAN communication and whats happening in LEAF are zero so it would be probably more errors then successes. But flashing new fw into can bridge every day for testing is not problem for me._

nzautomate, 2024-06-06:
[test binary.zip](https://github.com/user-attachments/files/15614368/test.binary.zip)
@malyjaponec you could try the code in this attachment. I put this code in my ZE0 JDM with 30kwh battery upgrade. I had it stop charging at 85% instead of 80. The code for estimating charge timers is totally different from dalas, as you found there is an issue with them. 

This is programmed to my needs, so 110V = 1.8kw charger, 220V = 3.6kw charger or thereabouts. 

There are many other fixes/easter eggs in it. It may suit your needs, or maybe not.

malyjaponec, 2024-06-06:
I think my need for this is kind of obsolete. With latest 4.x firmware the OVMS stopping charging quite reliable.

If you want me to test if the 80% stop works I can do it but just for sake of testing. I prefer simple working firmware now that does not have problem if the car is switched on quickly after switching off that had 2.xx version. It is much better with 4.xx. Charging I control with OVMS that can be scriptable depending on geolocation it itself sets proper maximum SOC. I do not time charging anymore because I very rarely charge to 100% and usually charge immediately after drive to 80. But if you want to test if timing works I can do it in some time. But if you have ZE0 with 30k it is same constellation like I have and it will probably work same way as on your car.

The only think I missed in 4.xx was power control of charging. Together with OVMS that already has some setting in app it would be nice if I can slow down DC to limit battery heating. But it is also quite out of my scope now. I stopped driving long distances with several charging sessions with Leaf as I have found sort of tesla vehicle.

dalathegreat, 2024-06-06:
Pull request is up for fixing this: https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/pull/33

Feel free to test @malyjaponec and others!

dalathegreat, 2024-06-07:
Issue confirmed fixed by testers! New software release is out! https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/releases/tag/v4.23

Closing this issue as solved ✅

closed 2024-06-07 by dalathegreat