dalathegreat/Nissan-LEAF-Battery-Upgrade #3: Add up all the nibbles in CSUM

InfantEudora, 2023-07-11:
Hi,
In your video: https://www.youtube.com/watch?v=oENNNfy5GSM you mention CSUM adds up all the nibbles, and add 2.

Mux's / Your code adds up only the nibbles from the first 7 bytes by the look of it, in `calc_sum4`
I was checking out some 1F2 messages from your log file, also this guy seems to send them: https://github.com/49thdiver/EV-Projects/blob/main/OBC/src/definitions.h
But they don't add up at all.

Most 1F2 messages from your log have 0 for the last nibble, so it works. But your DBC file has a bit in that nibble: "unknown - may indicate charging"

dalathegreat, 2023-07-11:
Not sure what is the question here?

calc_sum4 seems to work just fine, but does not fix anything. Running the code without recalculating the CSUM doesn't trigger any fault codes.

InfantEudora, 2023-07-11:
I guess the question is: Is the CSUM calculation used in that message? It seems to work without it right?

dalathegreat, 2024-05-26:
Yup, seems like no benefit. Closing this for now!

closed 2024-05-26 by dalathegreat