dalathegreat/Nissan-LEAF-Battery-Upgrade #4: Battery paired but lots of errors

gio-giorgio, 2023-08-02:
Hello,

Just installed the 2x gateway where a broken aftermarket solution was installed, and now I can read the range (281km) on the dashboard of the Leaf. But when using LeafSpy Pro I can see I still have some CAN bus errors, on the Brake and the HV Battery. 

Is it the ECU asking for a full reset to clear all the DTC faults, or is it guaranteed that some connection is not ok?

SOC does not show up at all on the dash, but on leafspy I can see 98.5%.

Thank you.

nzautomate, 2023-08-03:
What model of leaf is it installed in?

gio-giorgio, 2023-08-03:
It's a 2013 Nissan Leaf ZE0

gio-giorgio, 2023-08-03:
Ser No. 125L 001761, originally with a 24kwh and now has a 40kwh

gio-giorgio, 2023-08-03:
[image]

Here is a DTC readout with leafspy pro

gio-giorgio, 2023-08-03:
I can clear some faults but they come back after a while. Right now the main issues are the HV Can error, which could be related to the gateway.

[image]
[image]
[image]
[image]

nzautomate, 2023-08-03:
If this is a ZE0, several of those can fault codes are likely to show, and won't clear, but don't seem to have an effect on driving the car. P3196 is common on battery swaps, so is P30E0.

If yours is an AZE0, I understood that Dala had all the fault codes cleaned up as your swap is the same as he ran on his own car. 

I'd be looking to reflash and make sure you used the right file for your can bridge, and failing that, maybe your can bridge is faulty, especially if you have one of the AliExpress ones

gio-giorgio, 2023-08-03:
I reflashed it just now, afterwards I may try the other aliexpress gateway.

Anyway, the range shows up now but there is an I-Key fault there.

About the over voltage, I might discharge them a bit manually to see if that fixes it.

gio-giorgio, 2023-08-04:
An update:

We found that the 3rd party gateway I was using before had ignition also wired to the unit instead of being connected directly between the car and the battery.

So the issues I have been having could be explained by that. I would like to try to buy a harness that directly connects them and just modify it for the 2x CAN gateway, that could fix things.

Am I correct in assuming the connector of the 24kwh differs from the 40khw?

dalathegreat, 2023-08-15:
@gio-giorgio The connector on the battery is 22pin on ZE0, and 36pin on AZE0 and ZE1. Read the wiki: https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/wiki,

dalathegreat, 2023-09-14:
Closing this since it wasn't related to this CAN bridge originally

closed 2023-09-14 by dalathegreat