dalathegreat/Nissan-LEAF-Battery-Upgrade #6: No advanced features after 2 Port bridge installed

Jorindel53, 2023-09-07:
Hi, I installed a 2-port-budget bridge in my AZE0, but none of the advanced features work (I hav'nt tried Rapidgate Dodger, though).
I flashed the bridge successfully according to the instructions. It is connected to constant 12V and ground. The car has still it's original 24kWh battery, 79% SOH, 11bars. 
Have the advanced features been removed in the current firmware version? What else could be the reason? 
Will soon upgrade to a good 40kWh battery, which I already have.

nicstoun, 2023-09-13:
I think you need to flash this part to have it ..[https://github.com/dalathegreat/Nissan-Leaf-BatterySaver/blob/master/BatterySaver.c](url)

dalathegreat, 2023-09-14:
Hi @Jorindel53 That is correct, this software only focuses on getting the upgrade to work, nothing else. When I shipped the older version that had extra features, it constantly confused the users and caused people to accidentally activate extra features that they didn't want/need. This in turn led to emails asking for support, and supporting people everyday with using these extra features wasn't fun at all. So I decided to skip this entirely this time, for my own sanity!

closed 2023-09-14 by dalathegreat

dalathegreat, 2023-09-15:
Hi @nicstoun , Patreon supporters have access to the older software versions!

nicstoun, 2023-09-21:
> Hi @nicstoun , Patreon supporters have access to the older software versions!

Thank you :)