dalathegreat/Nissan-LEAF-Battery-Upgrade #7: battery percentage shows N/A

westsince2001, 2023-09-15:
Unable to display battery percentage after refreshing the latest hex。how to fix？
[image]

dalathegreat, 2023-09-15:
@westsince2001 What vehicle? (LEAF/env200)? What generation? (ZE0, AZE0, ZE1)? What file did you flash?

westsince2001, 2023-09-18:
> @westsince2001 What vehicle? (LEAF/env200)? What generation? (ZE0, AZE0, ZE1)? What file did you flash?

AZE0,MD5 (can-bridge-firmware.hex) = 04ad55d8089029b8834f2799d8125cc3

dalathegreat, 2023-09-18:
@westsince2001 try this file
[can-bridge-firmware.zip](https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/files/12649904/can-bridge-firmware.zip)

westsince2001, 2023-09-19:
@dalathegreat Tks,i'll try it! ^_^

closed 2023-09-26 by westsince2001