dalathegreat/Nissan-LEAF-Battery-Upgrade #8: Some permanent fault codes on after battery upgrade

dalathegreat, 2023-10-24:
Some error codes can be present after battery has been upgraded. These codes are as far as we know harmless, and doesn't affect how the vehicle behaves. They do not cause CheckEV light to be illuminated. Vehicles will pass inspection/MOT/katsastus with these active.

    U1000 (general CAN comm issue)
    P30E0 No CAN comm received from VCM
    P318D (general CAN comm issue)
    P318E No CAN comm received from VCM
    P3191 No CAN comm received from motor controller
    P3196 No CAN comm received from OBC
    P3198 No CAN comm received from A/C Auto Amp

maheshsachintha, 2023-10-24:
Received P31B9 code after install 30 kWh battery to ZE0. Why? In addition to that it shows full battery bars. But battery is 9 bar and SOH is 73%
[image]
[image]
[image]

maheshsachintha, 2023-10-24:
@dalathegreat In addition to that, any difference between open source 2 port can bridge firmware and your selling 2 port can bridge??

dalathegreat, 2023-10-24:
I created a new issue for the bars(https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/issues/9), and added P31B9 COMMUNICATION ERROR to the known error codes for ZE0 in this issue

I do not sell any CAN bridges anymore!

fearlesschicken, 2023-10-26:
I recently did a 2019 40kwh swap into a 2011 and have a bunch of these codes so this is really good info to know.
However, I also have P32DA-83 L-ION BAD CTLR and P3191-00 L-ION BAT CTLR Comm Err EVC-257.  Not sure if this is another one that appears on ze0s.
[image]

CrAzYDr1veR, 2023-11-10:
Hi first gen leaf sometimes the car will not work at all
[image]
[image]
[image]
[image]
[image]

CrAzYDr1veR, 2023-11-24:
mk1 with 62kw
[image]

dalathegreat, 2023-11-25:
> Hi first gen leaf the car shut down once while checking errors it has a 24kw nmc battery 

The battery upgrade CAN bridge is meant for upgrades, not for installation with 24kWh battery.

Oggioffshore, 2024-02-29:
Just upgraded 2012 Leaf from 24 to 30Kwh battery. Are getting EV fault and turtle mode /no power or throttle not responding. Max 35-40 km/h.. But cruise control can increase the speed though. Can't read any dtc.. Any suggestions? 

[image]

Oggioffshore, 2024-03-17:
Solved. User failure in leaf spy. Didn't know u had to check for service, dtc readout every time u start app.

jbenet1, 2024-04-23:
[image]

jbenet1, 2024-04-23:
Any help, please!🙏

dalathegreat, 2024-05-26:
@CrAzYDr1veR and @fearlesschicken 
Attempted fix is up in this PR: https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/pull/28

malyjaponec, 2024-05-27:
after update I have:
U1000-00 C0 EV/HEV  CAN Comm Circuit
P31B8-00 C0 EV/HEV CAN Error EVC-287
U1000-00 4E HV BATTERY CAN Comm Circuit
P3196-00 4E HV BATTERY CAN Error EVB-99
these can not be deleted. I seems after several restarts of car there are no new errors appearing. I think it in the past there were like two times or 3 times more errors.
Only one was added after restart:
P31C0-00 C0 EV/HEV TCU EVC-273 
probably due disconnected telemetric unit. 

I will try to DC charge once I find time if/what more errors appears. But I have only CCS chargers around so I will have to use adapter.

dalathegreat, 2024-06-05:
Happy to hear that the amount of errors went down with the latest update @malyjaponec , thanks for testing!

dalathegreat, 2024-06-10:
The last effort to get the final codes to go away is staging towards a stable version in pull request https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/pull/34 , soon we will have a completely error free ZE0 experience!

boydsta, 2025-04-23:
I get this error P0AA0-0B,
the car will not try the battery solnoids. unsure what to do next, I tested the battery solenoids and get ~33ohms and ~110 ohms for precharge one, Leaf sapy gets correct data from the battery.

30kw AZE0 adding a ZE1 40kw Battery

[image]

dalathegreat, 2025-04-23:
@boydsta Your issue is not related to the battery upgrade. The positive contactor is stuck/welded. See the MyNissanLEAF forum for more troubleshooting tips.