dalathegreat/Nissan-LEAF-Battery-Upgrade #9: 2011-2012 ZE0: Battery SOH bars stuck at 12

dalathegreat, 2023-10-24:
2-Port CAN-bridge. Shows full battery health bars. But battery should be a 9 bar and SOH is 73%
[image]
[image]
[image]

_Originally posted by @maheshsachintha in https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/issues/8#issuecomment-1777685051_

nzautomate, 2024-06-06:
I think I solved this - code worked for the bar level in the ZE0 I programmed it for. 
First a Look up table:
`uint8_t		bars_lut[16]						= {0,1,2,3,4,4,5,6,7,8,8,9,10,11,12,12};`

A variable to store the segments reported by the installed (AZE0 in my case) battery:
`volatile	uint8_t		capacity_segments_aze0  = 0;`

Then work with the two to report the segments to the ZE0 in "ZE0 formatting"
```
case 0x5BC:
...
if( My_Leaf == MY_LEAF_2011 )
...
if(frame.data[4] & 0x01) {capacity_segments_aze0 = ((frame.data[2] & 0xF0) >> 4); }
swap_5bc_full.LB_CAPSEG = bars_lut[capacity_segments_aze0];
```

The ellipsis represents unrelated code which is already part of the canbridge
I got the values for the Look up table from the can database files dala hosts - I assume they are right but I have not tested them.

dalathegreat, 2024-06-06:
@nzautomate I took those values from my AZE0, so I am not 100% those values work with ZE0. But thanks, I can open another PR with this also added!