dalathegreat/Nissan-LEAF-Battery-Upgrade #10: 2011-2012 ZE0: Bars are rescaled incorrectly when charge ends

dalathegreat, 2023-10-25:
Only affects ZE0.

When a charge is stopped, the vehicle will show 12/12 charge bars regardless of real SOC% if the amount of GIDs is over ~260. This is frustrating since it makes it hard to use the bars. Kms remaining still work, and leafspy shows correct SOC%.

This is due to bad coding in the 2011-2013 ZE0 VCM. No known software workaround. Here is how others have solved it:

-     Muxsan installs another CAN-bridge between cluster and VCM to make it better.
-     EVs Enhanced reprogram the VCM to make it behave correctly like on the AZE0

Oggioffshore, 2024-02-19:
Muxsan installs another CAN-bridge between cluster and VCM to make it better:  assume this canbridge requier different flash file/ programming than the flash file for the battery can bridge? Is there any threads around for cluster canbridge? 
tnx.

chadconway, 2024-03-20:
How does one reprogram the VCM? I'd like to acquire the tools and skills to do this.

nzautomate, 2024-05-20:
A ZE0 workaround is to raise the GIDS value when the car is turned on to the capacity of the pack 
`swap_5bc_remaining.LB_CAPR = (LB_FullCap_for_QC/0.8);`
After about a second, trigger the full charge flag in the 0x1DB message 
`frame.data[3] = (frame.data[3] | 0x10);// set the full charge flag to on `
leave that set for a second, then clear the charge full flag again, and then lastly pass through the battery reported Gids to the car. It was possible to almost hide all of this during the dash bootup graphics  - the battery briefly looked full, then looked corrected to the actual battery level. My code is messy and spead all over the place so I wont share the rest of the code snippets here - someone keen should write something that gives this effect.

dalathegreat, 2024-06-04:
Massive thanks @nzautomate , I opened a PR for this! https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/pull/30

dalathegreat, 2024-06-06:
This bug has finally been solved with v4.22 🥳 https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/releases/tag/v4.22

closed 2024-06-06 by dalathegreat