dalathegreat/Nissan-LEAF-Battery-Upgrade #15: 2 Port Can bride is not working even after successful message on programming

maheshsachintha, 2023-11-10:
2 Port Can bride is not working even after successful message on programming
[image]
[image]

uditha1980, 2023-11-10:
Why you say doesn't work....did u connect it to the can bus?...then what happened....
Need more detail of Ur issue to give an answer

maheshsachintha, 2023-11-12:
> Why you say doesn't work....did u connect it to the can bus?...then what happened.... Need more detail of Ur issue to give an answer

Yes. I connected it to car. But didn't work. 
Car Is 2013 AZE0 Car
Battery is 2018 40 kWh ZE1 battery

dalathegreat, 2023-11-12:
- How does it not work? 
- Are the battery info visible on the instrument cluster?
   - If not, the bridge itself might be broken 
- Any fault codes active on Leafspy?

uditha1980, 2023-11-13:
> > Why you say doesn't work....did u connect it to the can bus?...then what happened.... Need more detail of Ur issue to give an answer
> 
> Yes. I connected it to car. But didn't work. Car Is 2013 AZE0 Car Battery is 2018 40 kWh ZE1 battery

Problem with the low cost stlink programmer..
Use a genuine one may avoid most problems..
I compare the two files (before upload and the one in the MCU)...both are not same...but it worked for me...repeat the program session 2 or 3 times....it will ok...may be problem with buffers

maheshsachintha, 2023-11-17:
> * How does it not work?
> * Are the battery info visible on the instrument cluster?
>   
>   * If not, the bridge itself might be broken
> * Any fault codes active on Leafspy?

Used already paired car
Even initially not worked, it was worked after some programmed tries

maheshsachintha, 2023-11-17:
Any way to check successfulness program over can bus messages?

maheshsachintha, 2023-11-18:
I have programmed 8 number of can bridges. And compare srec file using STM32 ST-Utility. Got comparison (same) error as following. Still I haven’t plug this to car. I ll put update once I have checked with car
 
[image]

[image]

 
**Item 1**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 2**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 3**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 4**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 5**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 6**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 7**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154
**Item 8**
00000000	02010000	07060403	44330908
22116655	66774433	01AE4455	02BC0154

maheshsachintha, 2023-11-18:
> I have programmed 8 number of can bridges. And compare srec file using STM32 ST-Utility. Got comparison (same) error as following. Still I haven’t plug this to car. I ll put update once I have [...]

Any way to check this, without plugging to car

dalathegreat, 2023-11-19:
@maheshsachintha Flashing should only be done with BridgeFlasher.exe, not with STM32 ST-Link Utility.
The diff is due to the copy protection still being present (au8_lock)

You did not explain how the vehicle did not work. Please only flash with BridgeFlasher.exe, and provide the following info:

- How does it not work?
- Are the battery info visible on the instrument cluster?
   - If not, the bridge itself might be broken
- Any fault codes active on Leafspy?

Oggioffshore, 2024-02-22:
Leaf 2012, 24kwh upgrade to 30Kwh. Experienced error message no at link connected.. Cable was correctly installed. Also had to disable read out protection. Still didn't flash. After upgrade st link with new software it's worked... Just in case other struggling. Never done this earlier. Now I'll try connect it in the car, believe it will work even I have not changed battery? 
[image]
[image]
[image]
[image]
[image]
[image]

Oggioffshore, 2024-03-17:
Battery swop done, been working fine for a few weeks. Guess only drawback is the battery indicator and guess'o meter are working differently... Happy with 30Kwh an extended life for our 2012 Leaf.

fopoku2k2, 2024-03-17:
> Battery swop done, been working fine for a few weeks. Guess only drawback is the battery indicator and guess'o meter are working differently... Happy with 30Kwh an extended life for our 2012 Leaf.

Need to also do a swap on my 2012. How do I contact you for guidance??

Oggioffshore, 2024-03-17:
All you need to know are described in this forum. Just read.

dalathegreat, 2024-06-04:
Closing this, since it is not an issue with the code

closed 2024-06-04 by dalathegreat