dalathegreat/Nissan-LEAF-Battery-Upgrade #18: Errors

Smurfy4, 2023-11-27:
[image]
Had these issues since I've changed the battery in Aug 2023 did update on cambridge today 27/11/2023 and still on there any advice to get rid

dalathegreat, 2023-11-29:
@Smurfy4 
Can you provide more info on combination of parts?
JDM/EUDM/USDM vehicle?
Model year?
Generation?(ZE0,AZE0,ZE1)
Battery size?(24/30/40/62)

Smurfy4, 2023-11-30:
It's a 2013 ze0 eudm with 2019 40wkh

dalathegreat, 2023-12-04:
Thanks @Smurfy4 , so it seems like the 40kWh is not seeing all CAN messages that it would need to be error free. Someone with a ZE0 could try adding more CAN message sending, according to the log files I have over on https://github.com/dalathegreat/EV-CANlogs Lots of work to get it error free! But not sure if worth it, the car will function just fine with those errors present.

dalathegreat, 2024-06-04:
@Smurfy4 , latest version should fix this! [Available here: V4.21](https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/releases/tag/v4.21)

closed 2025-01-15 by dalathegreat