dalathegreat/Nissan-LEAF-Battery-Upgrade #21: 2011-2017 Not possible to quickcharge for over 60minutes

dalathegreat, 2024-01-01:
Noticed on AZE0, most likely also same on ZE0.
When quickcharging the charge will end abruptly when it reaches 60minutes. This is very noticeable on cold days, when quickcharging for instance a 62kWh battery. We need to look at the QC_Timer and see if this can be improved somehow.

[image]

dalathegreat, 2024-01-01:
Most likely the issue stems from this line of code: https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/blob/main/Software/CANBRIDGE-2port/source/Src/can-bridge-firmware.c#L399

patrickHall4, 2024-03-24:
Can I just say a quick thanks to Daniel for the great work  

So I have a 40kw 2018 with 251,000km  on the clock with a battery failure ! 

[image]

I replaced it with 2020 62kw battery Cleared the P3102 with Leafspy and spent 8 hours installing the 2 port canbridge ( most of that time was spent banging my head on a wall because  the dash would not read correctly)
!

[image]

however once i removed the canbridge and cleared the DTC's everything worked perfectly (almost) 

[image]

 until i drove 250km away for home and plugged into a fast charger with 20% battery only to get told i was at 100%  within 10 seconds and the session ended ! ( that was a long day !!!! )     

To get around the problem I reinstalled the canbridge and can now fast charge but there are lots of errors !
 Any help would  be appreciated !

nzautomate, 2024-06-08:
Do we know if this behaviour is not normal for the vehicle? My env200 which doesn't run this can bridge software stops after 60mins too

dalathegreat, 2024-06-09:
@nzautomate The 2018+ 40/62kWh vehicles can fastcharge for over 60 minutes on my Kempower charger
[image]

So unsure, it should maybe be possible to do the same for the upgraded vehicles? We just need to modify the minutes remaining to some sane value