dalathegreat/Nissan-LEAF-Battery-Upgrade #24: Fast charging is working!!!

patrickHall4, 2024-03-25:
Can I just say a quick thanks to Daniel for the great work  

So I have a 40kw 2018 with 251,000km  on the clock with a battery failure ! 

[image]

I replaced it with 2020 62kw battery Cleared the P3102 with Leafspy and spent 8 hours installing the 2 port canbridge ( most of that time was spent banging my head on a wall because  the dash would not read correctly)
!

[image]

however once i removed the canbridge and cleared the DTC's everything worked perfectly (almost) 

[image]

 until i drove 250km away for home and plugged into a fast charger with 20% battery only to get told i was at 100%  within 10 seconds and the session ended ! ( that was a long day !!!! )     

To get around the problem I reinstalled the canbridge and can now fast charge but there are lots of errors !
 Any help would  be appreciated !

dalathegreat, 2024-03-28:
The code is unfortunately meant for 2011-2017 LEAFs getting larger batteries. For 2018+ I have no solution unfortunately!

patrickHall4, 2024-04-07:
Hi just a quick update ! 
so Nissan called the car in for an update to fix a issue with the adaptive cruise control and my error message went away ! 
Maybe a note could to added with your instructions 
and thanks so much for all the hard work!

closed 2024-04-07 by dalathegreat