dalathegreat/Nissan-LEAF-Battery-Upgrade #25: 2011-2012 ZE0: Fastcharging ends with Check EV light

dalathegreat, 2024-03-27:
Cosmetic issue, after fastcharging completes, the CheckEV light will be illuminated. It goes away on its own after a power cycle (vehicle OFF, and ON again)

Help wanted, I was not able to solve this issue!

System1c, 2024-03-29:
This was noticed on my 2013 AZE0 even on a 24kWh pack (No Bridge, stock). It occurs one in every ~10 DCFC's and only when using a SETEC Chinese made charger. 

The DTC ends with VC-63, with the DTC starting with a U code.

Charging on ABB, Tritium and Siemens chargers have never caused this DTC. Probably caused by improper CHAdeMO implementation by charger manufacturer.

nzautomate, 2024-05-20:
I did solve this issue last year, these code snippets are related to the solve but I can't recall which one/combination was required. My old car is running all of this code so I don't think there is any harm in these being added for testing.

static		can_frame_t		AZE0_45E_message	= {.can_id = 0x45E, .can_dlc = 1, .data = {0x00}};
static		can_frame_t		AZE0_481_message	= {.can_id = 0x481, .can_dlc = 2, .data = {0x40,0x00}};
case 0x5B9:
			send_can(battery_can_bus, AZE0_45E_message); // 500ms
			send_can(battery_can_bus, AZE0_481_message); // 500ms
			if (My_Leaf == MY_LEAF_2011)
			{
				frame.can_dlc = 7;
				if(charging_state == CHARGING_SLOW)
					{
						frame.data[5] = 0xFF;
					}
				else
					{
						frame.data[5] = 0x4A;
					}
			
				frame.data[6] = 0x80;
				
			}
			
			break;

case 0x50A:
                     if(frame.can_dlc == 6)
                     {      //On ZE0 this message is 6 bytes long
                           My_Leaf = MY_LEAF_2011;
                           frame.can_dlc = 8;
                           frame.data[6] = 0x04;
                           frame.data[7] = 0x00;
                           if (frame.data[0] == 0x06)
                           {
                                  if (m50a_count > 10)
                                  {
                                         frame.data[0] = 0x00;
                                         m50a_count++;
                                  }
                                  else
                                  {
                                         
                                         frame.data[0] = 0x04;
                                  }
                                  
                           }
                           else
                           {
                                  m50a_count = 0;
                           }
                           
                     }

case 0x50B:

			VCM_WakeUpSleepCommand = (frame.data[3] & 0xC0) >> 6;
			
			if( My_Leaf == MY_LEAF_2011 )
			{
				CANMASK = (frame.data[2] & 0x04) >> 2;
				frame.can_dlc = 7;
				frame.data[6] = 0x00;
			}

			break;

dalathegreat, 2024-06-10:
Closing issue as resolved! 🙌

closed 2024-06-10 by dalathegreat