dalathegreat/Nissan-LEAF-Battery-Upgrade #26: Control DC charging current (feature)

malyjaponec, 2024-03-31:
There was such feature (overheating prevention) that enables slowing down charging to ~30 and ~20kW on DC.  It was controlled with manipulating the HVAC settings. But there is in OVMS feature that allows control car charging speed. I think these 2 things could work together?

I expect it tries to sends some CAN message that is not implemented for LEAF so it fails. Could be CAN bridge programmed to receive such CAN message from OVMS device and reduce charging speed to limit current during translation packets from battery to car, like ancient CAN bridge software did. 

I found it very useful (limitation of DC current). The car/battery itself has speed limitation when temperature raise to 56C in two or more steps. This works to prevent baking battery but it is applied too late so when it stops temperature raise between 56-57C it also reduce power. Car is then driving with 40... later 30 and if temperature gets near to 57 it goes to 20kW and shows turtle on dashboard. 
If the speed reduction is applied from begin of trip on each charging it would never get to the power limitation. There is the price of slower charging but later it does not give you power that has problem to drive up hill.

This current limitation could be designed that must be set on each charging (there is feature in OVMS app that can control it from 8 to 70A, but I hope the app will be soon new and some changes in number to set for example 100A would be possible) and it resets next time. Or other way is to use permanent value controlled byt OVMS settings. I am not expert on OVMS but there are some numbered variables that can be set by app or scripts. Maybe this variable is already there or there are some #x numbers that can be set by app and for example #7 is SOC limitation (if I remember well). Using these someone can set maximum power before trip and do not care each time.

_In meantime I am going to try swap to to very ancient CAN bridge sw that controls it via FAN speed because I found that if driving over 300km in row I need definitely slow down DC charging._

dalathegreat, 2025-06-16:
**Your AI solution will not work, and will break the battery.**

Use the old zipped .srec, and never experiment like that. This older version has all the features, but it is counted as not supported at the moment

[canbridge_v4.02.srec.zip](https://github.com/user-attachments/files/20759240/canbridge_v4.02.srec.zip)

dj-nitehawk, 2025-06-16:
thank you so much @dalathegreat 
i will try the v4.02 srec as a last resort.

i've just seen the following code on one of your youtube videos:

[image]

any chance that could work with the current code base?
if so, the only question would be under which CAN message frame byte 2 and 3 needs to be modified?
deepseek suggests `0x55B` 🫣

or am i completely off base here trying to patch the latest code?
is it somehow not possible to achieve with the latest code?

thank you for your time and support 🙏🙏🙏

dalathegreat, 2025-06-16:
Dude, AI is not the way to go here. Study the DBC file which has the CAN mappings. Do not rely on a hallucinating LLM for a safety critical machine.

dalathegreat, 2025-06-17:
Okay I think you should stop now. If you really are bent on trying to destroy your battery, go ahead.

You opened QC-CAN, which is the Chademo CAN bus. It has nothing to do with the battery.

Sorry for being blunt here, but fastcharging with modified CAN data is the fastest way to burn down the car. **Stop experimenting, use the provided .srec file**. If you mess up here, you will be allowing too much power into the battery, and the cells will thermally run away.

dj-nitehawk, 2025-06-18:
alright... message received, loud and clear.
thank you for preventing me from killing my car.
greatly appreciate the work you do @dalathegreat 
best of luck with your endeavors!!!

dj-nitehawk, 2025-06-22:
after a couple of days of research and sleepless nights, i have the following:
```c
    case 0x1DC:
      if (startup_counter_1DB > 100) // Startup complete?
      {
        if (charging_state == CHARGING_QUICK) // DC charging active?
        {
          if (can_bus == battery_can_bus) // From battery?
          {
            int qc_current = ((frame->data[2] & 0x0F) << 8) | frame->data[3];
            if (qc_current > 1000) // Over 50A?
            {
              qc_current = 1000; // Clamp to 50A (0.05A resolution)
              frame->data[2] = (frame->data[2] & 0xF0) | ((qc_current >> 8) & 0x0F);
              frame->data[3] = (uint8_t)(qc_current & 0xFF);
              calc_crc8(frame);
            }
          }
        }
      }
      break;
```
what do you say @dalathegreat? 3rd time's the charm? 😁
if this is wrong, i promise i won't bother you again... 🙏