dalathegreat/Nissan-LEAF-Battery-Upgrade #29: Will not DC fastcharge

multitron1, 2024-06-03:
Hi Dala.
First, thank you for the effort you have put into this project. I am a big fan.
I have done a upgrade on a 1. gen leaf  ZE0 with a-23 model 40kWh pack.  It works fine exept for the DC charging. It will not play along. I have never managed to fast charge after the swap. I am getting some fault codes: see screenshot. What will be the proper way to adress this, you think? I used V4.20 for the canbridge and followed your description.
[image]

dalathegreat, 2024-06-04:
Please try the latest release [available here: V4.21](https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/releases/tag/v4.21)

It reduces amount of fault codes active, making troubleshooting easier.

multitron1, 2024-06-22:
I Just installed the latest software and the car can now fastcharge. Fantastic,  thanks:-)

closed 2024-06-23 by dalathegreat