dalathegreat/Nissan-LEAF-Battery-Upgrade #31: SOH decreased after update from 4.20 to 4.21

malyjaponec, 2024-06-04:
After update from 4.20 to 4.21 I found that SOH changed from 86,6 to 76.9 %.

As the 30kWh battery should have aprox 356 gids fully charged and I have 312 gids it seems it is seems if we count SOH as percentage original capacity the number reported under v.4.20 was closer reality. 

I did not find it immediately, but after ABRP showed me that I can not drive where I normally drive as it now reports 23% degradation instead of 13%.

Remaining capacity looks the 4.21 reports same as 4.20.

dalathegreat, 2024-06-05:
Hi @malyjaponec 
This is most likely a real change. On my own 30kWh battery that I had in my last ZE0, **it was stuck at 91%SOH for over 3 years**. Seems like we finally solved enough error messages by sending the proper CAN messages towards the battery, so that the BMS is free to re-calculate SOH% again.

I would not worry if I were you. SOH% is known to go up and down on 24/30kWh packs. Try performing lots of 0-100% slowcharging sessions to help it re-calibrate.

Another reason I think this is not a bug, is due to the fact that we just pass thru SOH% without any modification.

Good luck with the re-balancing!

closed 2024-06-05 by dalathegreat

malyjaponec, 2024-06-05:
I bought this battery (made 2016) with leafspy info with 86,6 SOH.

I use it less then 2 years mostly driving between 70-80%. Only in every one or second month it drives longer distance where it needs like 15-20 minutes of charging on DC. I was recently trying to bake the battery with 500km challenge where we reached the temperature limit for turtle. 
I do not believe it removed 10% SOH in less than 2 years of using it, especially because I still can drive do Prague and have 30+% SOH there and ABRP was always almost perfect with prediction. Now when the car reports 23% degradation it shows me that i will get to Prague with 10-%. That is not happening in reality even if I try to keep maximal speed set in ABRP. The consumption is with every drive decreasing little bit in ABRP (after winter) so ABRP seems to be well calibrated regarding consumption. The only thing that recently changed was SOH that made the ABRP predictions very very conservative now.

By the way I will try to do 100%-0%-100% cycle soon. So we will see if it changes and how much. I will let you know. Some way.

nzautomate, 2024-06-07:
@malyjaponec , the frozen SOH value was the reason I took (a lot) of time working on this Canbridge code. 

The BMS actually can report the SOH in two ways through active polling. There is an Android app leafhacker which was able to do this and I did figure out how they did it. The car has a SOH which is continuously recalculating and when the car is turned off it should copy across to the place where leafspy is reading it from. It was this copy across which wasn't happening causing the SOH to freeze, however the BMS was aware that the capacity had dropped in some way. 

I found that we were running into much lower cell voltages and never hit turtle. I think we damaged our battery to at least some extent by getting to these low cell voltages which we were initially unaware of. The fix was put in as a fix to the DC fast charging check ev light but I was aware that the frozen SOH was also fixed with this.

malyjaponec, 2024-06-07:
During discharging 2 days ago I hit the turtle. It lights up when remaining capacity dropped from 0.6kWh do 0.2kWh. I have also voltages in moment of dying. The worst cell were about 2.7-2.8V and the best were 3V. I can see big difference between horizontal cells or cells on very edge of rear row/column and the cells that are in middle. It was over 0.3V. The battery definitely did some hard work. It is 8years old. 

As I expected after some driving and charging SOH changed very little. It went up about 0.3%. 

Regarding turtle I think it appears really at the point when the car starts to limit power under 30kW. I had reached power limit because over 45C in battery 2 or 3 times using this battery and because it was about 40kW limit it did not showed turtle. Once I reached 55C or more and power was limited to 20-30kW the turtle showed up even I had enough charge. The circles on the dashboard was like only 3 activated and also no regen. It was due to high temperature. So I think the turtle is not activated with SOC directly it is just a indication when the car has very limited power to tell you that you would not expect it will accelerate at all.

I think 2.7V in battery is not damaging. Getting there is just deep cycle that is damaging battery like every cycle. Little bit more then shallow cycle. I never got to this area when driving. I usually charge when car is at 30-40 if I can. If there is no charger I am very restless when get to 20%. Even at this SOC it is over 3V. It goes under 3V when there is like only 10% SOC or even less and normally I never gets to this voltage. I think temperature and time is most probable killer for my battery or leaf batteries. 2 charging in row and you see 40-50C and it stays in the well insulated pack (with air between cells and walls) for very long time.

nzautomate, 2024-06-07:
I agree 2.7V in itself isn't damaging. With the SOH not recalculating we had seen 2.7V per cell but not yet showing a very low battery warning, still 8-10km range estimate showing on the dash. I sold the car soon after getting the SOH to recalculate and on the one test I did, the VLBW was showing well before a 2.7V per cell value was seen. Before I was aware that we could get such low voltages without seeing the VLBW my wife would pretty regularly drive into VLBW as part of her daily driving, I have no evidence of what cell voltages we were getting to as she wouldn't monitor it via leaf spy.RegardsBrett McKenzie

nzautomate, 2024-06-07:
Also, through my testing I learned that the ZE0 would show turtle based only on a verly low gids (kWh capacity) value as you described (no limited current, faking a low gids value only with software). It may show turtle based on the current limiting also though I never tested that at all. RegardsBrett McKenzie