dalathegreat/Nissan-LEAF-Battery-Upgrade #35: ZE0 with heavily degraded pack will show 12/12bars too soon

dalathegreat, 2024-06-10:
The hardcoding of capacity remaining should be removed, and instead use a (Whan new capacity * SOH%). This will avoid showing 12/12 bars of range left when in fact it should be emptier. Not a massive issue, but could be done smarter.

nzautomate, 2024-06-10:
@dalathegreat , is there a reason you can't use the calculation I gave to estimate the full pack? It works for the 30kwh pack, and was quite accurate. Does the qc capacity reported by the battery not work the same in a 40 or 62 kWh pack? Qc capacity/0.8 (80wh/gid?)

nzautomate, 2024-06-10:
@dalathegreat had a quick look across the code, the 3port code is missing the 30kwh battery detection code. This is staying registered as a 24kwh battery on 3 port bridge. That is why the gids are not raised to the full pack level on startup

dalathegreat, 2024-06-11:
Not sure how you sample the QC Capacity, but massive thanks for the review! I added the 30kWh detection to 3-port, PR is up! https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade/pull/36