dalathegreat/Nissan-LEAF-Battery-Upgrade #37: Not possible to AC charge on ZE0 after update to 4.24

malyjaponec, 2024-06-28:
Yesterday I was testing some betas and then put into car 4.24 as betas does not bring anything good (reported). Car behaves well until I put it on AC. It clicks with contactors and after a second or two they are released and shows 3 blue LED on dashboard as it is charger. I have OVMS and I could see something like that SOC jumped up after inserting cable from 4x% to 8x% for a short moment. I have seen it 2 times. Sometimes is start charging with pressing release button on cable that car interprets probably as reinsertion of cable but it takes short time until it stops again. I disconnected OVMS to check if it is not caused byt it. Same result. I can not charger as car things that battery is full and stops charging or it even does not start.

It might be connected with some hacks to tune dashboard bars. 

After reprogramming 4.23 it charges on first try with OVMS connected, no SOC jumps observed.

dalathegreat, 2024-06-30:
Hi  @malyjaponec thanks for the feedback! I have removed the 4.24 as a stable release, the 4.23 is now marked the latest stable.

I unfortunately don't have much time to focus on getting these bugs fixed, but hopefully someone else might be able to debug it!

malyjaponec, 2024-06-30:
I realised that I commented you two marge requests and in one I did not find any issue (also no improvement) but I just switched on the car, checked dash and so one... I think it needs to test at lest short AC charge. I add it to testing for next time.

By the way car is perfectly drivable with 4.24, the problem came when wife returned next day back and I found that car is connected and not charging. 

After installing 4.24 I cleared several times to remove all old error and still did to not get clear list. There is screen from leafspy. 
[image]
Anyway it is the smallest amount of not erasable errors I ever had.