dalathegreat/Nissan-LEAF-Battery-Upgrade #39: Leaf 2017 40kWh -> 62kWh can problem

00gtw00, 2024-08-01:
Hello everyone. I have decided to upgrade the battery in my 2017 Leaf from 40kWh to 62kWh. The car charges with AC and drives without any issues, but it does not charge on fast DC chargers. It shows errors in EV/HEV U1009 and in HV BATTERY P30E0. I found a Nissan technical bulletin stating that the U1009 error appeared in Nissans with a 40kWh battery and that an update of the EV/HEV module software, also known as VCM, should be performed. However, fast charging worked with the 40kWh battery. Could the communication problem be caused by differences in the CAN protocol? Has anyone used a CAN BRIDGE with the new Nissan Leaf?

dalathegreat, 2024-08-01:
Which CAN bridge software version are you on?

00gtw00, 2024-08-01:
I haven't installed the CAN BRIDGE yet. I thought that in the LEAF ZE1, the battery could be replaced with a larger one without it. So, will the CAN BRIDGE be necessary in this case?

dalathegreat, 2024-08-01:
CAN bridge is needed for safe upgrade. Please install one.

closed 2024-08-01 by dalathegreat

00gtw00, 2024-08-01:
So, your firmware for the blue board won't work in this case? It only works for ZE0 and AZE0? 

If you'd like, we can work on this together.

marcucci-consulting, 2024-11-17:
Same issue with a US market 2018 ZE1 and 62kWh pack from a 2022, but only the P30E0. Haven't tried fast charging yet but I noticed the est charging times (except my single/preferred after key-off) are blank on my dash. We are an edge case but I'd like to sort this out. Haven't poked around the code much. It looks like the 40/62 has the same 8 byte msg for the MY_LEAF_xxxx determination... but @dalathegreat it sounds like you say the bridge won't work in the 2018? I have one in my 2015 I've swapped to 40kWh but not in my 2018 yet.

dalathegreat, 2024-11-17:
@marcucci-consulting That is correct. All the code in this repository is reformatting battery messages to suit either the ZE0 2011-2012 format, or the AZE0 2013-2017 format. There is no functionality at all for 2018+ LEAFs, so for this I still recommend getting the EVs Enhanced bridge.

If you want to develop a solution for this, we will need CAN logs from a 2018 40kWh, and compare the differences with a CAN log from a 2019+ 62kWh. I have log files available for the [62kWH here](https://github.com/dalathegreat/EV-CANlogs/tree/main/Nissan%20LEAF/62kWh)

DevThunder25, 2026-05-14:
My question to you is with this generation of leaf can I use that older firmware that you mentioned above to achieve the limitation on quick charge speed? My goal is to simply use the CAN bridge to limit the speed at which I charge at a quick charging station.