dalathegreat/Nissan-LEAF-Battery-Upgrade #43: ZE0 2011 24kwh-24kwh upgrade- not charging

Kontich, 2024-10-16:
Hello, 

I changed battery from 24kw SOH 50% (original, first generation, no PTC connector) to another 24kw SOH 82% (with PTC connector).
Battery pairing worked flawlessly, no turtle mode. Test drive did not indicate any issue.

After reaching home with 10km range I wanted to charge it, but nothing happen.
When I plugged in charging connector, no lights on front panel appear, only small beep is heard- no charging at all.

I tried multiple firmwares for 2channel CAN bridge but problem still persists

(Car is 2011 ZE0 white interior)
SHA256          EDA23950C991F48DB54768FCCE673104D86ABCF3FDF865403D600A843716BE9A       canbridge_1c2.srec
SHA256          39EF5DA6E0C6086A8F505CA0A8E2C42AD4409D9F27DA2BB9B5122C83037F42CE       canbridge_4.23.srec
SHA256          19CCB25D9F3C54D02536F04494353A13C7D231B0DDA3AEDAF23C1A3A0EB56EF6       canbridge_4.21beta.srec
SHA256          90C9BBDBE553E6DD2EC56A5C6EBD7A5FCC393BAFC057A89924A43A877F3C9418       canbridge_4.22.srec
SHA256          9A47F563C2DDBAA5F77FB83E8905F64A9F63589AC60356439EF8CA50A7C86731       canbridge_4.25beta2.srec

this is logs from LeafSpy:
[image]

I can't reset these error, it will appear over and over again.
Another problem is; temperature of battery pack is stuck on two blue bars, OVMS shows -2°C and it is not changing.
I think it is related to translation, because if plug in CAN wires directly without bridge, temperature is normal (around 10°C) and the bars jumps in middle of range on dashboard

But I found workaround how to charge the car:
turn ON the car
do the reset of EV/HEV 
turn OFF the car (quickly)
plug in charger (quickly)

OR

plug the charger
turn ON the car
do the reset of EV/HEV 
turn OFF the car (quickly)
press switch on the charging pistol / push reset charging timer on board (quickly)

than it will start charging, but this procedure is not wife-friendly.

Also I want to say big thank you Dala and community for big effort.

malyjaponec, 2024-10-18:
My OVMS shows completely crazy numbers in battery temperature sometimes. I have ZE0 with 30kWh. 

Now the app shows 11°C that seems to be ok as there is 12,5°C in garage and it is decreasing since yeasterday. Car was driven yesterday and then in afternoon charged. So I expect it should still have more then 11 that is UNDER the garage temperature.
So I did this list 
```
OVMS# me li temp
v.b.c.temp                               17.34,17.85,16.014°C
v.b.c.temp.alert                         0,0,0
v.b.c.temp.dev.max                       0.58,0.85,-1.12°C
v.b.c.temp.max                           17.34,17.85,16.014°C
v.b.c.temp.min                           10.914,11.016,10.914°C
v.b.p.temp.avg                           17.07°C
v.b.p.temp.max                           17.85°C
v.b.p.temp.min                           16.014°C
v.b.p.temp.stddev                        0.77°C
v.b.p.temp.stddev.max                    0.82°C
v.b.temp                                 11°C
v.c.12v.temp
v.c.temp
v.e.cabintemp                            18.6111°C
v.e.temp                                 13.5°C
v.g.temp
v.i.temp                                 22.7778°C
v.m.temp                                 22.2222°C
v.t.temp
xnl.bms.temp.int                         17,18,255,16,16,0°C
```
It looks 16 or 17°C is most probably correct battery temperature in the car now.  And then I switched on AC remotely to wake up car because OVMS sometimes is not able to read SOC or other stuff it the car is in deep sleep. The temperature of battery **jumped to -2°C**!

For complete information I run the list again when car was on doing preheating:
OVMS# me li temp
v.b.c.temp                               17.34,17.85,16.014°C
v.b.c.temp.alert                         0,0,0
v.b.c.temp.dev.max                       0.58,0.85,-1.12°C
v.b.c.temp.max                           17.34,17.85,16.014°C
v.b.c.temp.min                           10.914,11.016,10.914°C
v.b.p.temp.avg                           17.07°C
v.b.p.temp.max                           17.85°C
v.b.p.temp.min                           16.014°C
v.b.p.temp.stddev                        0.77°C
v.b.p.temp.stddev.max                    0.82°C
v.b.temp                                 -2°C
v.c.12v.temp
v.c.temp
v.e.cabintemp                            11.9444°C
v.e.temp                                 12°C
v.g.temp
v.i.temp                                 12.7778°C
v.m.temp                                 12.2222°C
v.t.temp
xnl.bms.temp.int                         17,18,255,16,16,0°C

What I see is that only v.b.temp changed. I think that OVMS is somehow do not know how to read battery temperature or use proper values to fill v.b.temp variable that is used for app display. When I did overheating test (I tried to drive until the car do not want to due to high battery temperature) the temperature in OVMS was like 45 but real battery was already baking almost at 60°C. 

I do not say anything about problem with charging on 24 to 24kWh upgrade but the temperature shown in OVMS is most probably  another issue.

dalathegreat, 2024-10-18:
This is an issue that should be handled on OVMS side and not on the battery upgrade side. The newer batteries have more complex 0x5C0 messages, which mess up the temperature reading.

Fix is to use the 0x7BB polled value for battery temperature (which is already read by OVMS)

malyjaponec, 2024-10-18:
By the way regarding dashboard. I just came to the car and it shows 2 temperature bars. According this:
[image]
There should be 5 bars if the battery has 16°C. So something is fishy in translating temperature. 

I got used to that SOC bargraf is sitting on full scale even I have charged to only 80% but I never checked temperature. I think it happens somewhere in recent time when I went to simplest firmware without all that fancy fan speed features. I have been trying overheating the car with very old software in bridge and it somehow worked there. I remember I reached red line :)

Anyway 2 bars does not mean -2 what was reported in the OVMS. It is like -10C... it that table is correct. Really translating data for ZE0 dashboard correctly is difficult task. It might be even impossible.

Kontich, 2024-10-20:
Fixed!
Problem was in 22-36 pin reduction  connector.
I bought this one 
`https://www.aliexpress.com/item/1005007015319732.html?spm=a2g0o.store_pc_home.0.0.3036901dFWcADF&gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=dd586388-7a74-4fb4-baf6-5faca6477aa4&_t=gps-id:pcStoreJustForYou,scm-url:1007.23125.137358.0,pvid:dd586388-7a74-4fb4-baf6-5faca6477aa4,tpp_buckets:668%232846%238111%231996&pdp_npi=4%40dis%21CZK%215508.04%215508.04%21%21%211694.00%211694.00%21%40211b65d417257843857764401e500b%2112000039079202895%21rec%21CZ%21%21ABXZ`

There was swaped one pin for charging. According to manual I resolder it to the correct place
Car is charging again and also I fixed temperature bars; I just reconnect TCU. 
[image]
I dont know what causes last error, but it is working. (I dont have PTC delete cap)

closed 2024-10-21 by Kontich