dalathegreat/Nissan-LEAF-Battery-Upgrade #45: AZE0 LEAF with battery swap not charging

halogen25, 2025-05-25:
I've just carried out a battery swap from a 24kwhr 2013 AZE0 with a 40kwhr from a wrecked 2024 LEAF with <3k miles, following carefully the instructions here and in Dala's youtube videos. After installation, while things look fine on the dash, the car will not charge. Multiplle EV/HEV DTCs are reported in LeafSpy that cannot be resolved.
Is this something others have encountered?  Or did I purchase a problematic battery?
Any suggestions would be helpful.

dalathegreat, 2025-05-25:
P31E1 is an interlock error. You did not seat one of the high voltage cables properly most likely

If you search for this code, you can find lots of references on the MyNissanLEAF.com discussion forum. Good luck with the troubleshooting!

halogen25, 2025-05-25:
Thank you Dala, for all that you do to help fellow LEAFers, and for your quick reply here.
After reading the helpful forums, I triple checked all of the connections, charged up the 12V battery, and cleared all of the DTCs with LeafSpy.  Now it is down to just 4 DTCs: C1A6E, C1A70, P31E1, and C118C. Others on the forum have had these errors, and they all point to troubles with the PTC connector. The grey wire on my PTC cable is fine, with no corrosion. Since the old battery worked OK, I figure this must a problem with PTC wiring inside the upgrade battery.  At least the car is still on jack stands, so I can drop the battery today!